timvideos/getting-started #30: [hdmi2usb] Port Linux to the lm32 CPU and support HDMI2USB firmware functionality

mithro, 2016-02-03:
# Brief explanation

The HDMI2USB gateware currently includes a [lm32 soft core](https://en.wikipedia.org/wiki/LatticeMico32). The misoc version has an MMU and should support running a full Linux kernel.

## Expected results

Linux booting on the HDMI2USB gateware.

# Detailed Explanation

Work on this was started by the MilkyMist / M-Labs people.

## Further reading

 * https://en.wikipedia.org/wiki/LatticeMico32
 * https://lwn.net/Articles/647949/
 * http://m-labs.hk/milkymist-wiki/wiki/index.php%3Ftitle=Milkymist_Linux_cheat_sheet.html
 * https://wiki.debian.org/HelmutGrohne/rebootstrap#lm32

# Knowledge Prerequisites

 * Linux Kernel knowledge
 * Strong C knowledge.

John-K, 2017-09-29:
Just dropping a note so that people know a good amount of work on this has been done in [linux-lm32](https://github.com/shenki/linux-lm32)

futaris, 2018-02-13:
http://j.mp/pre-fpga-lca2018

Linux isn't possible on lm32, due to bitrot. However we have it booting on or1k. 10/100 Ethernet is also working as of Jan 2018. See above instructions.

fallen, 2019-06-05:
Also, I might add that current lm32 Linux port, whatever its bitrot state, isn't using the MMU. It's uCLinux (noMMU).
MMU is experimental also.