timvideos/getting-started #31: [HDMI2USB] Rewrite the HDMI2USB JPEG Encoder in Migen/MiSoC

mithro, 2016-02-23:
# Brief explanation

The HDMI2USB firmware currently has a [JPEG encoder written in VHDL](https://github.com/timvideos/HDMI2USB-misoc-firmware/tree/master/gateware/encoder/vhdl). We would prefer it be written in [Migen / MiSoC](http://m-labs.hk/gateware.html) to allow better integration into the firmware.

## Expected results

The HDMI2USB MiSoC Firmware correctly generates JPEG output using new Migen / MiSoC JPEG encoder.

The new Migen / MiSoC JPEG encoder has a strong test suite.

# Detailed Explanation

The current JPEG encoder is based on [mkjpeg from opencores](http://opencores.org/project,mkjpeg) but has been slightly modified by @enjoy-digital and @ajitmathew. The current encoder has the following problems;
 * It is written in VHDL which means;
   * Isn't easy to simulate using FOSS tools,
   * An adapter between migen/misoc is needed,
 * It has been the cause of numerous bugs.
 * It doesn't have a strong test suite making modification hard.

Using Migen / MiSoC would allow a lot of better testing.

As there is an existing implementation of the encoder, it makes the conversion process a simpler. cfelton's JPEG encoder example might be useful source too.

## Further reading

 * [Replace VHDL jpeg encoder with Verilog version](https://github.com/timvideos/HDMI2USB-misoc-firmware/issues/47)
 * [Migen / MiSoC](http://m-labs.hk/gateware.html)
 * [cfelton's JPEG MyHDL repo](https://github.com/cfelton/test_jpeg)
 * [JPEG encoding](http://en.wikipedia.org/wiki/JPEG#JPEG_codec_example)
 * [mkjpeg from opencores](http://opencores.org/project,mkjpeg) 

# Contacts

 * **Potential Mentors:** @shenki @mithro @enjoy-digital 
 * **Mailing list:** [hdmi2usb@groups.google.com](https://groups.google.com/forum/#!forum/hdmi2usb)

adarsh1001, 2018-02-28:
I believe this particular problem was addressed in last year's GSoC also. How much work is left?

mithro, 2018-02-28:
@ishan98 was working on it last year. You can find a lot of his work here -> https://github.com/ishan98/litejpeg