timvideos/getting-started #32: [HDMI2USB #113]  Allow HDMI2USB devices to act as HDMI extenders via the Gigabit Ethernet port

mithro, 2016-02-23:
More technical details at [HDMI2USB-misoc-firmware#133: Allow HDMI2USB devices to act as HDMI extenders via the Gigabit Ethernet port](https://github.com/timvideos/HDMI2USB-misoc-firmware/issues/133)

# Brief explanation

Allow two HDMI2USB devices which are connected by Gigabit Ethernet to stream video from one device to the other.

# Detailed Explanation

It should be possible to stream the pixel data from one board to another board via Ethernet.

Gigabit Ethernet should be fast enough to transfer 720p60 video at raw pixel data (when using YUV encoding).

Both the [Numato Opsis](https://hdmi2usb.tv/numato-opsis/) and [Digilent Atlys](https://hdmi2usb.tv/digilent-atlys/) boards have Gigabit Ethernet interfaces. The [HDMI2USB-misoc-firmware](https://github.com/timvideos/HDMI2USB-misoc-firmware) already supports both these interfaces [using liteEth](https://github.com/enjoy-digital/liteeth).

As the two devices will be directly connected, you can use raw Ethernet frames rather than UDP/IP. It might be easier for debugging to support UDP/IP.

## Further reading

 * [HDMI2USB Website](https://hdmi2usb.tv)
 * [Migen / MiSoC](http://m-labs.hk/gateware.html)
 * [liteEth](https://github.com/enjoy-digital/liteeth).
 * [HDMI2USB-misoc-firmware](https://github.com/timvideos/HDMI2USB-misoc-firmware)
 * [hdmi2eth target](https://github.com/timvideos/HDMI2USB-misoc-firmware/blob/master/targets/atlys_hdmi2eth.py)

# Contacts

 * **Potential Mentors:** @mithro @enjoy-digital @shenki 
 * **Mailing list:** [hdmi2usb@groups.google.com](http://https://groups.google.com/forum/#!forum/hdmi2usb/)

T282, 2016-03-01:
Do I need some hardware to work in this project?

mithro, 2016-03-07:
Yes. While a large amount can be done in simulation, eventually it will need to be tested on real hardware.

mattookashyap, 2018-03-10:
I had some questions about the firmware design and where does the HDMI to Ethernet data converter fit in the hardware architecture of the HDMI2USB design.