timvideos/getting-started #33: [HDMI2USB] Get Milkymist "Video DJing" functionality on the Numato Opsis board

mithro, 2016-03-09:
# Get Milkymist "Video DJing" functionality on the Numato Opsis board

# Brief explanation

The Milkymist board provided a bunch of cool Video DJing output options (see following pictures), we want the same type of thing working on the Numato Opsis board.

[image]

## Expected results

The Numato Opsis when combined with a Milkymist Expansion board is able to reproduce all functionality of the original Milkymist device.

# Detailed Explanation

Very detailed explanation of the options around implementation can be found in the following Google Doc;
https://docs.google.com/document/d/1YyhCqTaUrI_vQjNGHJ34TsJCoDwpMzd2XcavAEllrUg/edit#

## Further reading

 * [HDMI2USB Website](https://hdmi2usb.tv)
 * [Migen / MiSoC](http://m-labs.hk/gateware.html)
 * [Numato Opsis board](https://crowdsupply.com/numato-lab/opsis)
 * [TOFE Milkymist Board](https://github.com/timvideos/HDMI2USB-TOFE-Milkymist)
 * [HDMI2USB MiSoC Firmware](https://github.com/timvideos/HDMI2USB-misoc-firmware/issues)

# Knowledge Prerequisites

 * C coding experience
 * Verilog experience very useful

# Contacts

 * **Potential Mentors:** @mithro @enjoy-digital @shenki 
 * **Mailing list:** [hdmi2usb@groups.google.com](http://https://groups.google.com/forum/#!forum/hdmi2usb/)

bradlaboe, 2016-03-20:
I am more of an collaborator/artist enabler and am an EXTREMELY poor C coder (I am dyslexic and have struggled for long years to attempt be a programmer)... how ever I am a long time systems architect and admin for film and video production... I am working on an  FPGA real time image synth (and have been not very successfully for a while) I acquired the Atlys board in late 2014 but but the project stalled.  After seeing this Numato Opsis board and the Milkymist IO board I ordered a set that and I am wanting for a shipping notification. My current plan is to use as the Atlys as a  front end switcher/frame-buffer/image preprocessor and the Opsis as the primary processor frame buffer.   I am currently getting my dev env chops up by enabling the Atlys board I have until I get the Opsis  ... let me know how I can help move this project.
Dyslexics Untie
-Bradley Laboe

mithro, 2016-03-20:
The capture firmware already runs on both the Opsis and Atlys board, so you can already get started with the HDMI2USB firmware on your Atlys board. Features developed with MiSoC will be portable to the Opsis.

Thus, the first step would be to get the development environment set up, you should follow the instructions at https://github.com/timvideos/HDMI2USB-misoc-firmware/tree/master/scripts - once you have that working there are a bunch of simple bugs you could try tackling in our issue tracker at https://github.com/timvideos/HDMI2USB-misoc-firmware

Once you have all that going, we can discuss more about how to get the MilkyMist stuff going. I think Option (2) from the Google Doc listed above is probably going to end up being the best solution.

bradlaboe, 2016-03-21:
Tim,
Created dev  system i5 4GB ram (will up to 8... sys max) 940 SSD running
Ubuntu14.04.1 64bit desk top Xilinx ISE 14.7 with ISE_WebPACK license

When I followed the recipe on the web page it failed to creat the build
directory... added the build directory by hand and worked the indicated
errors till the make gateware worked.... probably not worth documenting or
correcting as I should be one of the last users of it

I will do the jumpering and flashing tomrow

I look forward to using it with exiting opensource applications that accept
video in from your device from
http://signalculture.org/appclub.html#.Vu9hUuIrJD8  this will eliminate me
using camera re-scans for my input source.

Dyslexics Untie
Brad

mithro, 2016-03-21:
Which instructions were you following? The ones I linked (https://github.com/timvideos/HDMI2USB-misoc-firmware/tree/master/scripts)? If so, please do report any issues with the instructions.

I'm unsure why you said that you think you will be the last to follow them? The plan is to add the functionality to this repository.

bradlaboe, 2016-03-21:
Tim ... sorry I aid that just because they do not make the board any more
sorry my bad... but that being said yes that was the script that I followed
and when I did the  make gateware it it didn't make the build directory in
the /HDMI2UB-misoc-frimware tree and erroered out after the  3rd operation
 saying that it failed to make the build directory. I added the directory
by hand it worked with 1 more make needing to be done from an error  that
was a self explanatory.   Next time I will take better logs or I can try to
wipe it out and try to replicate the error if you'd like ... or go on to
jumpering and flashing tomrow

-Brad

mithro, 2016-03-21:
We plan to support the Atlys board for at least another year or more. I'm pretty sure it will be worth continuing to support the Atlys board for as long as the Opsis board (which uses the same FPGA) is in use. 

The Atlys was actually popular enough that after Digilent "discontinued it" they had to do another limited run of them. See https://store.digilentinc.com/atlys-spartan-6-fpga-trainer-board-limited-time-see-nexys-video/
> As of September 2015, Atlys is a legacy product and production is limited. The Nexys Video is recommended for new installations.

I think we also have roughly 8-10 of these boards in the whole team and they weren't cheap to purchase :).

bradlaboe, 2016-03-21:
Tim,
Agreed and it will be easy to add the extra HDMI, SDI or VGA as the
hardware already exists.... which is why in 2014 when I started my project
is why I picked it... cost and all of the above is why I kept it a the
front end even though I got the Opsis.... got it  the VmodMIB with it's 4
extra input would be great to have working making this a 6 X 2 HD switcher.
-b

mithro, 2016-03-21:
We should get the VmodMIB working (there is a bug open at https://github.com/timvideos/HDMI2USB-misoc-firmware/issues/31). I believe @enjoy-digital fried his Atlys board however?

I'm afraid you can't use all the extra VmodMIB connectors for extra HDMI inputs. The Spartan 6 only has 4 x Phase Lock Loops (PLL) and you need one 1 x PLL per HDMI input. You also need 1 x PLL for driving the memory and CPU. This means there is at most 1 x PLL free for usage with the VmodMIB which means only 1 more input is possible.

bradlaboe, 2016-03-21:
> timvideos/HDMI2USB-misoc-firmware#31

So that makes that a 1 HDMI and many Pmod in... still worth while... right
now the budget is tapped so I will get the VmodMIB in a month or so when I
can afford it.
-B

bradlaboe, 2016-03-22:
Tim, I am having 2 problems ... no matter what I do input 0 is not detected
nor does it detect video not sure weather it is the board or the software
and in video mode 11 it will sync with any monitors.

-B

bradlaboe, 2016-03-22:
Tim sorry I was tired last night... here's where I stand

1.) no matter what monitor I try mode 11 will not work from either output
port but all of the other modes work but with serious centering issues
2.) the input Labeled as input 1 can not accept in put and will not give
feed back to a source that is connected but the input labeled as input 0
works fine however the software thinks the numbers are reversed (IE: when I
plug in to the port labeled HDMI 0 in in your web diagram of the board it
sees it on input 1.   I wonder if this is a jumper issue. (if it is a fried
transceiver I am not sure what to do)

3.) I could use more test patterns split bars... stair steps..  circles for
checking aspect ratios etc how do I add these?
4.) We could use a standard as far as external test gear ... I will try to
hunt for a low cost alternative that has a more stable output than the pi
and will be playing with that any way to see what is the max rez intput I
can get from it.

Dyslexics Untie
Brad

mithro, 2016-03-23:
Regarding 1) and 2) - Are you using an Atlys board or an Opsis board? 
 * The Atlys does have a lot of jumpers which need to be set correctly for things to work. Use `video_matrix list` command to get help on how they should be configured.
 * The Opsis board has some [compatibility issues](https://github.com/timvideos/HDMI2USB-numato-opsis-hardware/issues/50) we are currently working on. The current work-around is to use active cables like [these ones with Redmere technology from Monoprice](http://www.monoprice.com/category?c_id=102&cp_id=10255&cs_id=1025503).

3) The test patterns are embedded in the C code running on the lm32 processor. See https://github.com/timvideos/HDMI2USB-misoc-firmware/blob/master/firmware/lm32/pattern.c and https://github.com/timvideos/HDMI2USB-misoc-firmware/blob/master/firmware/lm32/pattern.py - There is [open issue tracking this](https://github.com/timvideos/HDMI2USB-misoc-firmware/issues/185).

4) There is an [open issue](https://github.com/timvideos/HDMI2USB-misoc-firmware/issues/135) about building a better self test mode. The maximum resolution on the Atlys and Opsis boards are 1080p30 or 720p60