timvideos/getting-started #36: WIP: [MiSoC] Add support for RISC-V in MiSoC

mithro, 2016-04-14:
# Brief explanation

MiSoC currently supports the lm32 and mor1k architectures. It would be nice if it also support the RISC-V architecture.

## Expected results

MiSoC is able to use a CPU core which supports the RISC-V architecture as an option when building SoC components.

# Detailed Explanation

## RISC-V

RISC-V (pronounced "risk-five") is a new instruction set architecture (ISA) that was originally designed to support computer architecture research and education and is now set to become a standard open architecture for industry implementations under the governance of the RISC-V Foundation. RISC-V was originally developed in the Computer Science Division of the EECS Department at the University of California, Berkeley.

## Process

Steps would be;
 * Figure out the best RISC-V compatible core to use.
 * Figure out the toolchain needed.
 * Import into MiSoC.

Some potential options are;
 * https://github.com/cliffordwolf/picorv32 - PicoRV32 - A Size-Optimized RISC-V CPU
 * http://fpga.org/grvi-phalanx/ ?

## Further reading

 * https://github.com/m-labs/misoc
 * http://riscv.org/
 * http://www.lowrisc.org/
 * https://en.wikipedia.org/wiki/RISC-V

futaris, 2018-02-13:
https://twitter.com/q3k/status/955492603235053568
https://twitter.com/RohitK_Singh/status/955596375063801856

mithro, 2018-02-13:
@futaris - Yeap!

Need to get the toolchain into conda and it would also be good to get [VexRiscv](https://github.com/SpinalHDL/VexRiscv) imported for when you need a full RISC-V core.

mithro, 2018-10-16:
This is done now!

closed 2018-10-16 by mithro