timvideos/getting-started #38: [LiteX] Create a litescope based "Integrated Bit Error Ratio Tester" (iBERT) clone

mithro, 2017-02-28:
# [LiteX]  Create a litescope based "Integrated Bit Error Ratio Tester" (iBERT) clone

# Brief explanation

Xilinx has a [logicore](https://www.xilinx.com/ipcenter/ip_license/ip_licensing.htm) called iBERT for doing testing of error rates on high speed channels. The task is to create a similar tool based on LiteX and LiteScope.

## Expected results

What the expected outcome from the project should be.

# Detailed Explanation

This project has three parts;

 * Data sequence generators + checkers. These generate given bit data stream, then after transmission and receiving check that the bit data stream is correct. 

 * Data channel wrappers. These give you a common interface to controlling the parameters of a channel used in transmission and receiving. For simple data channels this might just provide clock control. For more advanced channels, like the high speed transceivers, this provides things like controlling parameters like pre-emphasis, equalisation, etc.

 * Host computer Control GUI / Console. This gives a nice interface for controlling all the parameters and seeing the results of various tests. This is the Xilinx iBERT Console -> [image]

## Further reading

 * [LiteX](https://github.com/enjoy-digital/litex)
 * [LiteScope](https://github.com/enjoy-digital/litescope)
 * [ChipScope Pro Tutorial: Using an IBERT Core with ChipScope Pro Analyzer](https://www.xilinx.com/support/documentation/sw_manuals/xilinx13_3/ug811_ChipScopeUsingIBERTwithAnalyzer.pdf)

# Knowledge Prerequisites

 * List of
 * what knowledge is required
 * to complete this project.

# Contacts

 * **Potential Mentors:** @mithro, @enjoy-digital 
 * **Mailing list:** [hdmi2usb@groups.google.com](http://https://groups.google.com/forum/#!forum/hdmi2usb/)

akhil-123, 2018-02-13:
I would like to work on this project. Any tips to follow?

anant3110, 2018-03-17:
I want to work on this issue. I might have access to a vertex 7 board from my institute. Would that work?