timvideos/getting-started #40: [LiteX] Improve support for project IceStorm (Lattice ICE40 + OpenFPGA toolchain)

mithro, 2017-02-28:
# [LiteX] Improve support for project IceStorm (Lattice ICE40 + OpenFPGA toolchain)

# Brief explanation

The first fully FOSS toolchain for FPGA development targets the Lattice ICE40 FPGA. LiteX has some support for this toolchain but it hasn't been very tested.

## Expected results

LiteX has good support for the IceStorm work flow and ICE40 based development boards. At least the LM32 and RISC-V soft CPUs should work and many of the peripherals should work.

# Detailed Explanation

 * TODO.

## Further reading

 * [IceStorm Toolchain](http://www.clifford.at/icestorm/)
 * [WhiteQuark's - Implementing a simple SoC in Migen](https://lab.whitequark.org/notes/2016-10-19/implementing-a-simple-soc-in-migen/) which uses IceStorm.
 * [icoBoard - FPGA based IO board for RPi](http://icoboard.org/)

# Knowledge Prerequisites

 * List of
 * what knowledge is required
 * to complete this project.

# Contacts

 * **Potential Mentors:** @{{github mentor username}}
 * **Mailing list:** [timvideos@groups.google.com](http://https://groups.google.com/forum/#!forum/timvideos/)

cr1901, 2017-09-18:
I'm the one who actually wrote the toolchain file for IceStorm. What would you like added :)?

mithro, 2017-09-24:
I think the steps would be;

 - [ ] Get yosys + IceStorm toolchain into [our conda repo](https://github.com/timvideos/conda-hdmi2usb-packages)
 - [ ] Add popular ICE40 boards into LiteX
 - [ ] Get simple `BaseSoC` using lm32 booting on the Ice40 boards.
 - [ ] Get SPI flash working (enable BIOS + firmware running from SPI flash)
 - [ ] Look at getting MicroPython running on the boards.
 - [ ] Have a bunch of demos which flash LEDs, etc.
 - [ ] Look at getting SDRAM working with ICE40

Dr4s1l, 2017-09-25:
Any improvment about micropython integration and LiteX compatibility?

mithro, 2018-10-16:
@cr1901 Is working on this....

mithro, 2018-10-16:
I'm also working on using Yosys as a frontend to Vivado here - https://github.com/enjoy-digital/litex/pull/114

mithro, 2019-07-29:
This has been done!

closed 2019-07-29 by mithro