timvideos/getting-started #41: [LiteX] QEmu simulation of a LiteX generated SoC

mithro, 2017-03-01:
# [LiteX] QEmu simulation of a LiteX generated SoC

More technical details at [Issue #86: Get lm32 firmware running under qemu to enable testing without hardware on HDMI2USB-misoc-firmware repo](https://github.com/timvideos/HDMI2USB-misoc-firmware/issues/86)

# Brief explanation

Currently to do firmware development you need access to FPGA hardware. Using QEmu it should be possible to enable developers to develop and improve the firmware without hardware.

## Expected results

People are able to write new features for the [HDMI2USB firmware](https://github.com/timvideos/HDMI2USB-misoc-firmware/tree/master/firmware) and develop [MicroPython for FPGAs](https://github.com/upy-fpga) using QEmu.

# Detailed Explanation

TBD.

## Further reading

 * [Link to a PDF document](http://abc/abc.pdf)
 * [Link to a some HTML page](http://abc/abc.html)

# Knowledge Prerequisites

 * List of
 * what knowledge is required
 * to complete this project.

# Contacts

 * **Potential Mentors:** @shenki 
 * **Mailing list:** [hdmi2usb@groups.google.com](http://https://groups.google.com/forum/#!forum/hdmi2usb/)

John-K, 2017-09-29:
Just dropping a note so that people know a good amount of work on this has been done in [qemu-litex](https://github.com/timvideos/qemu-litex/)

CarlFK, 2018-02-16:
mithro (IRC):  some basic stuff can now be done without FPGA hardware.