timvideos/getting-started #45: [SymbiFlow] Add support for Spartan 6 parts

mithro, 2018-02-12:
# [SymbiFlow #10] Add support for Spartan 6 parts

More technical details at [SymbiFlow Idea #10](https://github.com/SymbiFlow/ideas/issues/10)

# Brief explanation

Spartan 6 is a hugely popular part which would be awesome to have support for in open source FPGA toolchain - SymbiFlow

## Expected results

 * Documentation of the bitstream for Spartan 6 parts.
 * Support for using Spartan 6 in the SymbiFlow.

# Detailed Explanation

[SymbiFlow](https://symbiflow.github.io/) will be a FOSS Verilog-to-Bitstream FPGA synthesis flow for Xilinx 7-Series FPGAs and iCE40. SymbiFlow currently only supports Xilinx Series 7 parts and the Lattice iCE40 parts. 

While the Spartan 6 has mostly be superseded by the Artix 7 and Spartan 7 there are still a **huge** number of boards out there with Spartan 6 parts. Due to its huge popularity it will be a long time until the part is no longer in use (people still start new designs with Spartan 3!).

To make it even more important Spartan 6 designs you still have to use ISE which is significantly worse then Vivado in many ways. Having an non-ISE toolchain for that would be **awesome**.

Spartan 6 is used heavily by a number of open source projects;
 * The Novena Open Laptop - https://www.crowdsupply.com/sutajio-kosagi/novena 
 * TimVideos HDMI2USB Project - https://hdmi2usb.tv

## Further reading

 * [Link to a PDF document](http://abc/abc.pdf)
 * [Link to a some HTML page](http://abc/abc.html)

# Knowledge Prerequisites

 * List of
 * what knowledge is required
 * to complete this project.

fanoush, 2019-11-05:
There is also the amazingly cheap Pano Logic G2 (and G1 with Spartan 3E)

BTW there is interesting paper https://pdfs.semanticscholar.org/64c7/a8f1dc7b513aca6eea387d9c319c0b358506.pdf (3rd hit for https://www.google.com/search?q=open+source+bitstream+xilinx) that discusses methods of merging pregenerated combinations of basic bitstream files to get desired functionality. Using the method to generate desired bitsream looks more complicated than reverse engineering it but as a procedure/methodology to understand the bitstream it is a good read.

DaAwesomeP, 2021-02-08:
@mithro would you mind updating the Knowledge Prerequisites section? I would like to help out if I may, but I am really unsure what I would be able to contribute without knowing the tasks required (or what I need to learn).